GaloisInc/smaccmpilot-stm32f4 #9: smaccm-datalink: sometimes fail to receive frames in unit tests

pchickey, 2015-04-19:
I've found some problems where, nondeterministically, both the frame loopback and commsec loopback tests fail due to missing frames sent from the client (PC).

As best as I can tell, I think there are bugs with either the UART driver receive missing bytes in a way we cannot detect (currently we look at the overrun flag in hardware) or the PC-side client not getting every byte out onto the serial port.

One way I've tried to track down this bug is adding a debug variable that counts how many fbo bytes arrive at the uart driver. A successful two-frame receive and loopback will show 4 fbos, but sometimes the exact same code with the exact same initial condition will only get 3 - or less frequently 2, or 1 fbos.

I suspect it is the PC-side client failing to write properly, because when the tower-side UART driver receives and loops back a frame, there are never any corrupted bytes inside a frame. It seems to just miss the start of some frames.

I have encountered this bug under mac os. It would be good to get reports of whether the behavior persists elsewhere (with other ftdi cables, other operating systems, etc). To run the frame loopback unit test, inside `smaccm-datalink`, you should build the test firmware with `make smaccm-datalink-ivory-frameloopback-test`, load the firmware on a board, and then use the test client `cabal run smaccm-datalink-native-frameloopback-client -- --serial=/dev/cu.usbserial-A9I91FBB --baud=115200 --debug`

We could also see if using some other way of writing between haskell and the serial port resolves the problem, e.g. switching the haskell app to talk to stdin/stdout and using socat to connect to the serial port

pchickey, 2015-05-15:
Jamey and I believe this issue is resolved.
- There might have been bugs in serial port input/output on mac, and definitely were bugs on linux. these were fixed by dropping the `serial` library (which, due to a goof on my part, we were actually using the very ancient version 0.1 of, because that version did not specify a `base` constraint) and replacing the tty related code with a direct use of the `unix` lib: https://github.com/GaloisInc/smaccmpilot-stm32f4/commit/f41ceb7cdf58d49087874f271da2271e0a91ceae
- There was a bug in the commsec client due to the `gec` library initializing the aes-gcm counters at 0, whereas the ivory implementation expects them to be initialized at 1. https://github.com/GaloisInc/gec/commit/b234b184cedfdf05ebeab95be893d1175c635df4

closed 2015-05-15 by pchickey

TomMD, 2015-05-17:
Sorry about that, but in that topic: shouldn't the loss of a single packet, even the very first packet, be easy to recover from?  Honestly I'm surprised it was even noticed.

pchickey, 2015-05-17:
Well, we notice it because our unit tests expected a valid response for every request - a reasonable invariant in a test.

The GIDL attribute protocol does not (yet) support retrying if a request does not get a response. So, users of smaccm-comm-client do, currently, have a pretty bad failure mode when a packet is lost - the request HTTP connection idles for 20 seconds until the snap server times out and returns an error.

We should work on those things, but lately we've just been trying to get basic functionality out the door.