GaloisInc/smaccmpilot-stm32f4 #18: eChronos build fixes

leepike, 2016-09-30:
* Eliminate the need for RAMSES_PATH.mk and PRJ_CMD.mk in the build.
   But maybe add an warning if the path doesn't exist.
* Figure out the status of and integrate the AADL bootloader stuff (addresses, debug).
* Eliminate the Makefile warnings in `src/smaccm-flight/platform-fmu24/standalone_flight_echronos` --- e.g., the `mv` errors.

leepike, 2016-09-30:
Also,

* Add the `image.px4` target to the Makefile in `standalone_flight_echronos`, and make sure the required Python scripts are copied in.

podhrmic, 2016-12-16:
Now it fails on a circularity check.  @acfoltzer and @leepike know what needs to happen

podhrmic, 2017-03-02:
Closing since it is already resolved in master.

closed 2017-03-02 by podhrmic