GaloisInc/smaccmpilot-stm32f4 #23: Implement altitude estimator

podhrmic, 2016-12-16:


acfoltzer, 2017-01-05:
Do we want to derotate the lidar and sonar altitudes as well? It doesn't look like paparazzi does, but since we'll need the rotation state estimate anyway, perhaps we should?

acfoltzer, 2017-01-05:
Is the baro offset still useful even if we're never trying to measure it? Does the state eventually update enough through other measurements that the offset becomes accurate?

podhrmic, 2017-01-06:
Yes we do want to do the derotation, because in practice the measured and derotated altitudes differ in up to 15%. Here is the paparazzi code (it is bundled inside the lidar driver): https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/modules/lidar/lidar_lite.c#L144 

And indeed similar for the sonar.

I think we still want to know the baro offset - because we use it when we get a baro measurement, see here: https://github.com/paparazzi/paparazzi/blob/master/sw/airborne/subsystems/ins/vf_extended_float.c#L185

Without the offset the baro measurements would be less accurate.

acfoltzer, 2017-01-10:
Okay, so this is implemented on the `px4flow` branch. I realize now how the baro offset fits into the filter, but now I'm wondering more about the accel bias. The accel output that we're putting into the altitude estimator has already been calibrated, so maybe we could just drop this field from the state vector?

podhrmic, 2017-01-11:
hmm, good point. For the time being I would keep it as is though, and if we really have accel measurements without bias, the `accel_bias` state will converge to zero. I suspect the bias will be different for z-axis and maybe not fully removed in the attitude estimation module.

closed 2017-02-17 by acfoltzer