GaloisInc/smaccmpilot-stm32f4 #29: Change calibration for accelerometer

acfoltzer, 2017-01-20:
Rather than subtracting a g out of the accelerometer, we should figure out the bias when it is static and flat, and subtract those bits out of the raw register data we read from the MPU6000. As it is, we're subtracting 9.8 from the Z acceleration, even when that axis is no longer down!

acfoltzer, 2017-01-20:
And remember to subtract 9.8 out from the derotated accel in the altitude estimator.

acfoltzer, 2017-02-02:
@podhrmic I have merged this in, before I close the ticket though could you write up some documentation about how to do the accelerometer calibration so other folks can do it on their vehicles?

podhrmic, 2017-03-02:
Documentation in progress, so far it still waits for the new attitude estimator - once finished I ll change the documentation accordingly. First changes are already there http://smaccmpilot.org/software/preflight.html

closed 2017-03-02 by podhrmic