GaloisInc/smaccmpilot-stm32f4 #33: GCS: only poll endpoints when relevant views are visible

acfoltzer, 2017-02-17:
For example, when the PFD tab is selected and the camera target isn't visible, we shouldn't poll the camera target endpoint. This will require keeping some extra state in the `Model` but will let us keep a higher refresh rate. (cc @agacek)

acfoltzer, 2017-03-02:
See cb151bf4e02fb7aa8975c7f2389be3989e14f30b

It was a long walk around this block in order to separate out the polling loops for different endpoints, but the network performance should be a lot better now as a result. Note that when we have the camera target selected, we poll both it and the packed status, since the packed status still might be populating the control law display.

@agacek the `elm.js.min` file is updated in this commit, but if you want to make further changes using Elm yourself, you'll have to rebuild `smaccm-comm-schema` with an up-to-date `gidl`.

closed 2017-03-02 by acfoltzer

agacek, 2017-03-02:
This fixes the flickering issue I was having on the PFD display. The flickering persists on the camera view display (probably due to updating 2 things at once?).

acfoltzer, 2017-03-02:
By flickering, you mean the errors popping up at the bottom? Does increasing the timeout help?

agacek, 2017-03-02:
Yes, by flickering I mean the timeout messages popping up at the bottom.
Increasing the timeout makes them less frequent, but they still persist.

acfoltzer, 2017-03-02:
I cut the knot: b4f7759005916435189b4f785ad433f6b4a54694

agacek, 2017-03-03:
It works great now. Thanks.