GaloisInc/smaccmpilot-stm32f4 #35: Alternate 'hacker' GCS

agacek, 2017-03-03:
For the phase 3 demo we would like the GCS to have a different 'skin' for the attacker. Is there any easy way we could make the background black or something and change the other text to still be readable? We want  to convey that all the same information is there, but it's somehow a different system.

acfoltzer, 2017-03-03:
Should be able to do this fairly easily.

acfoltzer, 2017-03-03:
With the latest master, try http://localhost:8080/elm2.html

It's not the greatest-looking theme, but is at least distinct from the normal one. I can probably go in and tweak aspects of it if you have requests.

agacek, 2017-03-04:
Looks good. I'm sure we'll tweak things in Virgina based on how readable
things are on the big projectors.

-Andrew

closed 2017-03-04 by acfoltzer